ShawMerlin/NES-Power-Module-Redesign #1: BOM parts are back ordered,out of stock, or expensive.

acfsas, 2022-06-29:
I was in the process of ordering the parts for the Rev D Through Hole RF as Composite Video Only Edition version of this board and I quickly found the parts listed are back ordered, out of stock, or expensive. I used the BOM found here https://github.com/ShawMerlin/NES-Power-Board/blob/master/Bill_of_Materials_NESPowerPCB_TH_BOM_Mar2022.xlsx
Image for reference:
[image]

ShawMerlin, 2022-06-30:
Thank you for letting me know, that BOM is several months old.  Let me update it and find in stock parts and try and reduce cost.
-Merlin Shaw

ShawMerlin, 2022-06-30:
Not sure where the OctoPart is getting some of these values, I wonder if my format for the BOM doesn't match their specs.  Your picture shows the 9 Pin Gen 2 Port as water pipe costing $3000.  if you click the link, it takes it to Mouser where the part costs $3.29...  Let me do some checks on this Octopart and see if I can get a list that will play nicely with it's algorthm.

https://www.mouser.com/ProductDetail/CUI-Devices/MD-90SM?qs=WyjlAZoYn53AvrHguf2LdQ%3D%3D

ShawMerlin, 2022-06-30:
I have entered all the parts needed for the Composite Video Only Board on Octopart's BOM Generator.  You can access it at the below link.  I will work on getting the BOM excel files updated to play nicely with it.  Puts all parts needed for 1 board at $12.
Everything will be at Mouser as that is the part supplier I use but there are some parts at Digikey as well.

https://octopart.com/bom-tool/vQfUDfYZ

Let me know if there are any further concerns with this part list.
Thanks,
-Merlin Shaw

ShawMerlin, 2022-06-30:
I have used the Octopart's BOM Tool to generate a web link to the parts needed for the Fork of Spink's Rev D Composite Video Only board.  I will work later tonight on updating the excel but have deleted it at this time.  Use this link to get the parts needed, total comes to $12 for 1 board.  Everything can be obtained on Mouser as that is the part supplier I use, Digikey also holds most of the parts.

https://octopart.com/bom-tool/vQfUDfYZ

Let me know if you have any further concerns with the BOM's
Thanks,
-Merlin Shaw

acfsas, 2022-06-30:
So why is Mouser confusing one part for another? On my Mouser shopping cart I have
[image]
instead of 
[image]
Octopart seems to know what specific parts are listed on the BOM while either Mouser or DigiKey doesn't? IDK?

acfsas, 2022-06-30:
[image]
My Mouser list

ShawMerlin, 2022-06-30:
Those Manufacture numbers do not even match what I put in Octoparts.  not sure why it's pulling the wrong parts.  I will do some more digging.

ShawMerlin, 2022-06-30:
On the OctoPart Link, when I click Mouser Buy Now, it adds all the parts correctly.  Perhaps you still had older items in your cart.  Go clear your entire Mouser Cart and then click the Buy Now again from OcotPart.

ShawMerlin, 2022-06-30:
Those extra parts are amounting to all the extra costs that is shown on that picture.  Having it cleared, it should be around $21 after shipping.

acfsas, 2022-06-30:
Alright... 
[image]
Why is duplicating the parts?

ShawMerlin, 2022-06-30:
Did you clear the whole cart or just the first 4 items?

That looks correct for ordering enough parts for 2 boards.  
Let me find a replacement for the LED light.  What color do you want for the power LED?

acfsas, 2022-06-30:
I cleared the whole cart. I don't mind the color- what is the led used for?

ShawMerlin, 2022-06-30:
I updated the BOM Tool on Octopart to include an in-stock LED.
You can delete the backordered ones and order this part.  It's a blue LED.

[150080BS75000](https://www.mouser.com/ProductDetail/Wurth-Elektronik/150080BS75000?qs=2kOmHSv6VfT2tBo3UPVqAQ%3D%3D)

acfsas, 2022-06-30:
Thanks dude

ShawMerlin, 2022-06-30:
Anytime, I will have the other BOM's up tonight.

closed 2022-06-30 by acfsas