ShawMerlin/NES-Power-Module-Redesign #2: What happened to the Classic RCA DIY THT Edition?

NovenTheHero, 2023-02-01:
I went to go get the files and apparently they have been deleted.

ShawMerlin, 2023-02-01:
After it tested good, I noticed later that the inductor was getting far too hot during longer sessions so I pulled the files in order to come up with a better solution.

That was a while ago so I apologize about not having anything.  I will see if I can work on something this weekend for it.

Thanks,
-Merlin Shaw

closed 2023-02-01 by ShawMerlin

NovenTheHero, 2023-02-01:
Oh awesome thanks man, I was just curious =)

ShawMerlin, 2023-02-14:
@NovenTheHero 

Just wanted to let you know that I have uploaded the new gerbers and BOM for the DIY Board.

https://github.com/ShawMerlin/NES-Power-Module-Redesign/tree/main/Classic%20RCA%20DIY%20THT%20Edition

Let me know if you have any questions on it!!

Thanks,
-Merlin Shaw

NovenTheHero, 2023-02-15:
> @NovenTheHero
> 
> Just wanted to let you know that I have uploaded the new gerbers and BOM for the DIY Board.
> 
> https://github.com/ShawMerlin/NES-Power-Module-Redesign/tree/main/Classic%20RCA%20DIY%20THT%20Edition
> 
> Let me know if you have any questions on it!!
> 
> Thanks, -Merlin Shaw

Oh awesome thanks a ton!