ShawMerlin/NES-Power-Module-Redesign #5: NESRGB Companion board: No power

Scyllinice, 2023-11-12:
Hello, 

I built up one of the aforementioned boards and can't seem to get the NES powered on. I did not forget the ground connection to the main board.

I followed the 9 volts around and it comes up everywhere I think it should.

Do you have any suggestions on what to look for? 

I should note that the power led on the board also doesn't light up. Should it light up when the board isn't soldered to the NES?

Scyllinice, 2023-11-13:
I built up another one just to make sure I didn't do anything wrong.

It worked for a couple of power cycles and then abruptly stopped working.

This is the 2.1d Genesis version

ShawMerlin, 2023-11-13:
@Scyllinice 

I hope you saw my email on the other thread but let me address some things here.

If the board is not soldered to the NES, then the power LED will not light up unless the top 2 pins on the 5 pin intererconnect are shorted across, you can use a pair of needle nose plyers to short these 2 pads in order to test the board.

These top 2 pins are labelled, 9 volt out and 9 volt in.  The out pin goes to the power switch on the front and then it comes back and feeds into the 9v in pad which powers it when the power switch is depressed.  So by sorting out those 2 pads, it simulates the power button being pressed.

At this point, you should get a power led light as long as the solder jumper is connected that is next to the LED.

Where did you have these boards made from?  I noticed from JLCPCB that some of the boards had solder paste still on them so when the chip heated up, it shorted out the input and output on the voltage regulator.  Before is sent boards out to anyone, I would do a 2 Amp load test on the board for a few hours just to make sure there we not manufacturing defects like this.

If I were to redesign the board (which I do not have time for anymore)  I would use a TE Voltage Regulator that would have a bigger footprint to allow for assembly houses to have greater tolerances on the requirements.

Let me know what you find out when shorting out those pins and if you have any other quetions, i will do my best to help you out however I can.

[image]

Thanks,
-Merlin Shaw

Scyllinice, 2023-11-13:
Hello,

I did see the other issue. I'm going to check to see if I can get something to happen on the board that worked for a few cycles.

I did get mine made from JLCPCB. I didn't get any of the boards assembled there. I assembled it by hand so if there are assembly problems, it would be all on me.

I'll report back if I find anything interesting.

Thanks for making the board and taking the time to answer the issue!

Scyllinice, 2023-11-25:
I finally got my NESRGB installed and everything is working great. Thanks!

closed 2023-11-25 by Scyllinice