ShawMerlin/NES-Power-Module-Redesign #6: How to power the NESRGB board? (Genesis 2 RGB version)

jdfr228, 2023-11-12:
I built one of these myself, the Genesis 2 version, and got power on the console (power LED lighting up) but seemingly no power to the NESRGB board.

A regular NESRGB install in a frontloader has you hook up a separate voltage regulator to feed it with cleaner 5V power.  Since this board is already redesigned from the ground up I figured the whole point is the 5V coming off of it is clean.  So I read 5V on the PPU and an odd 1.6V or so on the NESRGB, and decided to jumper J3 to connect PPU 5V (end pin at the top left) to the NESRGB's 5V line.

It very briefly lit up and then everything turned off.  Now I get no power LED on the system, even with the J3 unjumpered and nothing on the NESRGB board.  I forget where my voltages were at afterwards but clearly I blew something.

So how is power supposed to be hooked up to the NESRGB board with this installation?  I didn't see any instructions about this on this github which I found odd, and especially sucks if it did cause me to blow something expensive to replace (kind of doubting the NES has handy fuses installed...)

Scyllinice, 2023-11-13:
Obviously not the author, but I have a couple of questions:

1). Was it the 2.1d version of the board?
2). How long did it work before it stopped working?

I build up a couple of 2.1d version boards and neither worked more than a few power cycles before failing. The first one didn't get as far as making the power button LED light up.

I wonder if there's something electrically unsound about that version.

jdfr228, 2023-11-13:
Yup, 2.1d. I swapped some parts out for availability reasons but followed their datasheets.

Stopped working as soon as I jumpered the pad to give the NESRGB power (J3). Will give the original power board a try once I'm back at it.

How did you power the NESRGB? J3, extra regulator, a wire to 5V, anything?

Scyllinice, 2023-11-13:
> Yup, 2.1d. I swapped some parts out for availability reasons but followed their datasheets.
>
> Stopped working as soon as I jumpered the pad to give the NESRGB power (J3). Will give the original power [...]

My NESRGB arrives this week, so my builds failed without it being involved.

I built following the BOM (I didn't notice the BOM included parts that don't exist on the board). I didn't swap any parts.

I have seen one YouTube video where one of these boards was paired with the NESRGB and it was powered via the j3 jumper. No problems there, but I don't think it was the 2.1d version of the board.

jdfr228, 2023-11-13:
Interesting - definitely curious if you can get things working with the NESRGB, even if it's briefly. Would appreciate if you'd measure the voltage on the NESRGB's 5V line before jumpering J3 as well.

I'll also look sometime this week to compare the difference between 1.9 and 2.1d. Maybe something will jump out.

Scyllinice, 2023-11-13:
I will do so. 

I know the 2.1d version rearranged things, but most parts are the same. The standout to me was that the 2.1d version uses a different bridge rectifier. If I wasn't already frustrated with the outcome so far and I needed more components, I'd order the previous bridge rectifier part and see if things play out differently. I just can't justify ordering a single part.

ShawMerlin, 2023-11-13:
@Scyllinice - @jdfr228 

I am sorry to hear your experiences have been poor on this board.  If you want to send me a private email, I will get some information from you both and send you boards that I have tested here myself and that are working.

We did update the bridge rectifier on the new version as it could handle more amps, the older one was getting pretty hot and we felt better about having it be over spec'd to handle all loads.

As long as the top pin on the 5 interconnect pins is showing 9-12v DC then the bridge rectifier wouldn't be the issue.
if you have 9V - 12v showing on the 2nd pin down on the 5 interconnect pins, this is the output from the switch and means it's feeding 9v's into the voltage regulator.

If you have 5 volts out on the 3rd pin down from the 5 pin interconnect but only 1.6 volts showing on the NESRGB.  I don't know what would have caused that and I personally have not seen that before.   Perhaps put a jumper from the 3rd pin down which is 5v out to the 5v in pad on the NESRGB to bypass the NES Main board.  Just a thought.

Please let me know if you have any questions and I will do my best to answer them.

Thanks,
-Merlin Shaw

Scyllinice, 2023-11-13:
> @Scyllinice - @jdfr228
>
> I am sorry to hear your experiences have been poor on this board. If you want to send me a private email, I will get some information from you both and send you boards that I [...]

If you've got some built up and working already, I'd certainly pay you for one and send you back the 3 remaining unpopulated boards I have from JLCPCB if you want.

ShawMerlin, 2023-11-13:
Just shoot me an email to merlin.shaw@gmail.com and I will get you taken care of.

jdfr228, 2023-11-14:
> @Scyllinice - @jdfr228
>
> I am sorry to hear your experiences have been poor on this board. If you want to send me a private email, I will get some information from you both and send you boards that I [...]

I'm willing to keep trying things to see what's going on, unless you have multiple extra Genesis 2 RGB boards lying around.

I'll put the original power board back in to make sure nothing's fried.  Then try the new while checking voltages and making sure everything matches what you said it should.

Can you confirm that jumper J3 on the NESRGB shouldn't be bridged?  You make it sound like the NESRGB should be getting 5V on its own, though I'm not sure where it would be pulling it from?  Since the original schematic has you connect the replacement regulator to the 5V pad on it, I don't know where it would pull 5V without J3 or a 5V wire?

ShawMerlin, 2023-11-14:
@jdfr228 - Send me an email to merlin.shaw@gmail.com and I will get you taken care of

ShawMerlin, 2023-11-14:
@jdfr228 - On my NESRGB, I have J3 soldered so it pulls the 5V from the Main NES Board.  The Main NES Board gets it power from Pin #3 of the 5 Pin Interconnect so it should all work.

If you are getting less than 5v on J3 then I would suggest not bridging that point and bring the 5v directly from Pin 3 with a jumper wire so that it's going directly to the RGBNES, this bypasses the route through the NES Main Board.

Send me an email and I will get you a few boards that I have made.

Thanks,
-Merlin Shaw

jdfr228, 2023-11-14:
Ah okay, probably want to ping @Scyllinice as well.

I had a seemingly stable 5V on the PPU pin J3 connects to before soldering J3, so I figured that was the trick to get proper 5V to the NESRGB, but maybe whatever that odd 1.6V was caused issues and overloaded that line.  Will look into it more when I'm back home at my soldering station.

ShawMerlin, 2023-11-14:
I am also in communication with @Scyllinice.  

I am not sure why it would voltage would be dropped to 1.6v, especially if you have 5v on Pin #3 on the Power Board.  This one is labeled 5v Out.

[image]

jdfr228, 2023-11-14:
The weird 1.6V was before I soldered J3. So like the NESRGB was picking up an odd voltage from somewhere. So maybe it partially shorted the 5V line to ground, or 6.6V was fed to things and fried something.

Hopefully nothing was permanently damaged but the possibility is there and I can't imagine the NES was designed super well for over voltage protection.

Scyllinice, 2023-11-25:
I finally got my NESRGB installed. I didn't have any weird voltages before closing J3. Everything is working great for me.

Did you board survive in the end?

jdfr228, 2023-11-25:
My board appears to be fine, the NES powers on and voltages seem okay. I ended up disconnecting one wire at a time to check where that voltage was coming from, and I guess I had a short on the controller wires at the bottom right?

What's stopping me at the moment is actually having Genesis 2 cables to test with, my usual supplier is out of commission, and the composite output still doesn't work. I will be swapping the composite and audio connectors with ones in the kit ShawMerlin kindly shipped out since I bought the wrong mounting type, so hopefully they work after that, though I connected a wire manually that should have done the trick...

Suppose there's a remaining chance I damaged something but hopefully not...

Scyllinice, 2023-11-25:
My composite output also doesn't work with the NESRGB board installed. It worked when it was just the PPU in the socket. I suspect that's just normal operation unless we both ended up doing something wrong :).

ShawMerlin, 2023-11-25:
With the NESRGB Board, the internal composite will no longer work.  This
has to do with the palette option from the RGBNES Board.
For example, if you put the NESRGB Board to manual pallets and then take
all solder joints off the 4 pads so it's back to how the board comes.  The
Internal Componsite will work.

But when any Palettes are applied from the RGBNES Board, it will not output
internal composite anymore, you will have to use the RGBNES Composite Output

I hope that makes sense. I know it's not the most clear on how that works.

Scyllinice, 2023-11-25:
@ShawMerlin 

I want to clear something up:

I just killed one of these boards with a bog standard official NES power supply. Some component can't handle the output from the adapter. I checked the output before I plugged it in and it was a little over 10v without load which is normal.

Should we expect it to work at that voltage? 

Off to build yet another one :(

jdfr228, 2023-11-25:
> With the NESRGB Board, the internal composite will no longer work. This has to do with the palette option from the RGBNES Board. For example, if you put the NESRGB Board to manual pallets and then [...]

Ah, that makes sense then, I was in fact trying the internal composite instead of NESRGB composite.

I would think sound would have worked though regardless? Will have to double check how that's wired as well I guess.

For the BL8033 @Scyllinice the spec sheet says 4.5V-16V input so you should be fine there, depending on what's coming off that 4-pin input chip (what's that called, a buck regulator? Rectifier?). I have replaced the BL8033 on mine though since I couldn't find the BL8033 in stock, if I confirm things are working maybe you'd want to try out an alternative? The only circuit change is an additional cap on the existing unpopulated pad for the part I bought.

Scyllinice, 2023-11-25:
Yeah, it's odd to me that it went dead. I swapped to a Triad 9V power supply and everything is working again. I'm not going to tempt fate on this build, but I might on the next one.

Scyllinice, 2023-11-26:
@jdfr228 Looks like I'm going to need your suggested alternative.

The BL8033 went short again, even though I did not put more than 9V in.

ShawMerlin, 2023-11-27:
The only thing in the schematic for the MP1471 that is different is it uses a 24.9K resistor while the current one uses a 10K resistor.  This is on component R4 from the diagram below.  You might be able to get the new part to work by replacing the R4 resistor with a 24.9K one.

[image]

ShawMerlin, 2023-11-27:
Also, the LSCS Buck Converter has a 100k resistor on R7 that is not shown on the MP1471.  You can remove that resistor and just bridge the pads together.

Let me know if this works for you as I think the MP1471 would be better to use instead of the original LSCS Part.
Thank you for all the input on this.

jdfr228, 2023-11-27:
Ah, good thing I asked for a double-check. I did indeed swap R4 out for a 24.9K one, I just forgot about that. So an extra cap and one resistor swap to change parts.

ShawMerlin, 2023-11-27:
Does everything work properly with that new voltage regulator used?

jdfr228, 2023-11-27:
Personally I hadn't used the other one, it's been MP1471 from the start on mine. I haven't needed to open your board up yet Shaw and will hopefully be able to pass the favor along to someone else who needs one.

My main holdups are cables, since I use VGA connectors for all my RGBS stuff, and a scaler, since I actually moved recently and don't have my OSSC with me, and am waiting patiently on the RetroTink 4K to launch. I may have to go grab the OSSC just in the interest of testing things, though I need to swap around composite video and csync signals to confirm my NESRGB install is working via composite at least.

Scyllinice, 2023-11-27:
If the MP1471 is working, then it stands to reason that the suggested newer MP1477 part would also work with a new board revision that switches the pinout around. It looks like pins 1 and 3 switched places as well as pins 4 and 6.

ShawMerlin, 2023-11-27:
> Oh, nevermind. It's because the MP1477 only comes in SOT563. I misread.
>
> SOT23 alternatives at least have the same pinout.

The tagline not recommended for new designs doesn't bother me too much, as
long as there is plenty of stock, things go years and years still being
available to purchase after going EOL.

I think that disclaimer is more for people that are going to mass produce
thousands and thousands of a design.  At that point, it might not be wise
to use an EOL Part.
But for us hobbyist that are going to use a dozen.  I see no reason not to
use those in new designs.  If that makes sense.

Thanks,
-Merlin Shaw

jdfr228, 2023-11-28:
Grabbed the OSSC to test and my NESRGB install is fully working now with the power board with the MP1471 and two circuit tweaks (extra cap and resistor value swap).

Can't say my situation was ever as bad as yours @Scyllinice but should I test various power cycles or leaving it on overnight or something?  Seems stable here but I don't plan on playing NES much this week (don't have the OSSC remote and its current settings didn't look great...)

Scyllinice, 2023-11-28:
If you can power it back up tomorrow with no issues, then you're in good shape. That'll be better than me.

It'll be good data to have.

jdfr228, 2024-01-06:
> If you can power it back up tomorrow with no issues, then you're in good shape. That'll be better than me.
> 
> It'll be good data to have.

Late reply, but I'm currently playing through Bionic Commando on it with my new Tink 4K.  Looks good and no power issues so far.  I'll let you know if anything changes, but the MP1471 seems fine with the resistor value change and extra cap.  Looks quite clean to me, and I'm not even using super high-quality cables.

ShawMerlin, 2024-02-27:
Closing this thread but if anyone has a purchased board that has failed, please reach out to my email.  If you would like a replacement, we can work on a discount that would cover my costs.

closed 2024-02-27 by ShawMerlin