ShawMerlin/NES-Power-Module-Redesign #7: Board Production query (Genesis 2 RGB version)...

gingamonkey13, 2024-01-01:
Hi - Firstly, thanks for sharing your designs here - A great solution to not hacking old consoles to pieces!

I'm in the process of bringing my old NES into the 21st century, and am attempting to get a 2.1d board produced at JLCPCB. 

They've just emailed me with the following issue - As a fairly new hobbyist in this field (*it's the first time I'm attempting to get a board produced myself!), any help would be much appreciated..

> Hello,
> 
> Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.
> 
> As shown below,comparing the drill diameter the drawing chart, it seems that the holes on the drill layer have been compensated, is the file used for production before? Usually,The drilling data in the original file are not integer, but now the drilling data in the file are integer, so we are confused.
> 
> Please help to confirm A or B?
> 
> A:If the file is production file, we fabricate as per this gerber without further compensation and optimization
> 
> B: If the gerber is original file, that means we will add +0.05 for NPTH and +0.15 for PTH, + 0.015mm trace width when we making production file..(Making clearance for the data of gerber file )

[image]

ShawMerlin, 2024-01-01:
That is strange because I have used that file a lot at JLCPCB.  If you
could send me the gerbers you have.  I would like to confirm them with what
I am using.

Also, I am about the release a new V3.0 of the power boards that will be a
better design so it might be worth waiting a few weeks for me to be able to
test out this new design.

Biggest change will be using a TE Voltage regulator and such better full
bridge rectifier.

Thanks,
-Merlin Shaw

gingamonkey13, 2024-01-01:
Thanks for the reply Merlin - I've attached the zip I sent through to JLCPCB (but it's just the same, untouched version from your RGB - Genesis 9 Pin Edition / Version 2.1d folder).

Good news on the v3.0 - I think I'll put a pause on it until you're done with that one. 

All the best (and a Happy New Year!),
Neil

[NES_RGB_V2-1d-Gerbers.zip](https://github.com/ShawMerlin/NES-Power-Module-Redesign/files/13804063/NES_RGB_V2-1d-Gerbers.zip)

ShawMerlin, 2024-01-17:
Just wanted to give you an update.

I am testing on these new boards but so far, they are running at 5.04v with
a full 1.5 amp load on them.  Clean and Stable power which is fantastic!!
https://github.com/ShawMerlin/NES-Power-Module-Redesign/blob/main/RGB%20-%20Genesis%209%20Pin%20Edition/readme.md

Thanks,
-Merlin Shaw

gingamonkey13, 2024-01-27:
Hey Merlin - Hope all is well with the new boards? 
Just wondering if you're any closer to releasing the gerber files, etc for the new version?

I'd already purchased most of the additional parts for the 2.1d version (from Mouser, Digikey, Amazon, etc) until I ran into the issue above with getting the board manufactured , so after your recommendation, I decided to hold out for the v3.0 board. Hopefully I can still use a lot of the parts I purchased with your new v3. 

Thanks!

ShawMerlin, 2024-01-27:
Testing the 3.0 boards has been going good but there is some issues with
the GNDing from all previous boards that I need to work out.
With this new discovery, I would not recommend any of the previous models.

It should only be a few more weeks until I can get everything vetted and
independent review on the new design.
Since you have already purchased things, let me send you some of the new
designed boards for cost, I will pay for the shipping.

Just let me know how many you need and I will put you on a list, I want to
give those that have put money into the older boards a chance to get a new
one for cost.

Thanks,
-Merlin Shaw

gingamonkey13, 2024-01-28:
Oh, really appreciate that Merlin - Thanks!

I'd originally ordered enough parts to make up 5 boards (*the minimum number I could get manufactured at JLCPCB). re. postage.... I insist on covering all costs, as imo it should be at our own risk that we attempt to use these boards that you have designed and kindly shared with the community. I wouldn't want you to have the burden of any expense!

Not sure how I share my details with you - Just let me know what to do when you're ready.

Thanks again! 
-Neil

ShawMerlin, 2024-01-28:
My email is ***@***.***
Shoot me a message and I will put your email on the list for replacements.

Thanks,
-Merlin Shaw

ShawMerlin, 2024-01-28:
You can contact me on my Geek Island Facebook page as it seems to not allow email on here.
https://www.facebook.com/GeekIslandGaming/

closed 2024-11-04 by ShawMerlin