ShawMerlin/NES-Power-Module-Redesign #8: Genesis 9Pin 2.d

victor7376, 2024-01-11:
Hi,

I'm making the above mentioned board & ordered most of the parts, but I've come across an issue between two documents.

On DOCS_POS_NES_PowerBoard_RGB_V2.1d.csv it states R1 is 680 ohm resistor

But on DOCS_BOM_NES_PowerBoard_RGB_V2.1d.csv it states its a 2.2k ohm resistor

Any advice which one it is please??

Cheers

ShawMerlin, 2024-01-11:
R1 is just the resistor for the Power On LED so the value really isn't a big deal.  I believe I changed it later to 2.2K so it wasn't so bright.

However, I am about to release a new RGB Module that I feel has significant improvements so you might want to hold off until the new version is released.

victor7376, 2024-01-11:
Cheers for that, I'll keep an eye on it.  I would hold off but I've already got 10 pcb's lol.

I did struggle wih getting some of the parts though but hopefully should have at least one board working next week when some orders arrive from China & USA.

ShawMerlin, 2024-01-11:
That was one of the reasons for the new design.  This time all parts will
be available from Digikey/Mouser.

Thanks,
-Merlin Shaw

victor7376, 2024-01-11:
perfect !!

Might put another order in lol

ShawMerlin, 2024-01-17:
I am testing on these new boards but so far, they are running at 5.04v with
a full 1.5 amp load on them.  Clean and Stable power which is fantastic!!

https://github.com/ShawMerlin/NES-Power-Module-Redesign/blob/main/RGB%20-%20Genesis%209%20Pin%20Edition/readme.md

Thanks,
-Merlin Shaw

closed 2024-11-04 by ShawMerlin