ShawMerlin/NES-Power-Module-Redesign #9: R5 and R6 at wrong location? (Classic Composite v2.1e)

bitdo1, 2024-01-16:
I tried loading the gerber, BOM, and POS files into https://jlcpcb.com/ and the position of R5 and R6 are not in the correct location.

Other minor things:
- J3, J4, J5 are missing from POS (the 3 RCA connectors)
- C11 50uP 50V cap has no inventory.  There were a couple 47uP 50V caps with same 0805 package to choose from.
- C8 220uF cap only had 5 in stock and I couldn't find an alternative with same dimensions

[image]

I saw in previous issue you are working on a new board so I'll hold off on this for now.

ShawMerlin, 2024-01-16:
Thank you for the feedback.  I will be removing this design once the new one is available but I appreciate you finding these.

The newer design (which I am currently testing) will have easier to source parts that can be bought at places like DigiKey/Mouser but also assembly at JLCPCB.

For an alternative to C8, you can use LSCS C2757945

I didn't include the Jacks in the POS as I just did those myself instead of having it assembled.  I didn't feel it was worth the cost of those connectors and soldering compared to just doing it myself.  I can include that in later releases.

Thanks again,
-Merlin Shaw

bitdo1, 2024-01-17:
Just read the "NES Revival RGBNES Companion Power Module" page.  If I am not doing NESRGB, can the 3.0 board be used as a replacement power supply with legacy composite video/internal audio?  Maybe just needs a jumper from J1 video to J8 comp.

ShawMerlin, 2024-01-17:
The NESRGB "Companion" Power Board can do internal audio/video.  These are
selected with the slide switches that are on the board.

closed 2024-01-17 by bitdo1