ShawMerlin/NES-Power-Module-Redesign #11: RAD2X Support?

Nibell, 2024-01-28:
Hi, i was wondering if it supports and working with RAD2X cables for Genesis 2?
[https://www.retrogamingcables.co.uk/sega/mega-drive-2/SEGA-MEGA-DRIVE-2-GENESIS-2-32X-HD-RAD2X-HDMI-480P-CABLE](https://www.retrogamingcables.co.uk/sega/mega-drive-2/SEGA-MEGA-DRIVE-2-GENESIS-2-32X-HD-RAD2X-HDMI-480P-CABLE)
Or it only works with rgb cables?

ShawMerlin, 2024-01-28:
Yes, this board will work with the Rad2x.  However a new version is being tested right now so I would not recommend using any of the older versions.

I do have one of the new 3.0 boards on my eBay if you needed one now.

Thanks,
-Merlin Shaw

Nibell, 2024-01-30:
> Yes, this board will work with the Rad2x. However a new version is being tested right now so I would not recommend using any of the older versions.
> 
> I do have one of the new 3.0 boards on my eBay if you needed one now.
> 
> Thanks, -Merlin Shaw
Thx, i will look in to it :smile:

closed 2024-01-30 by Nibell