ShawMerlin/NES-Power-Module-Redesign #14: Missing Schema for Version 2.1e for the Classic Composite Edition

fschuetz, 2024-02-07:
Hi

In the folder `NES-Power-Module-Redesign/Classic Composite Edition/Version 2.1e` the schema for KiCAD is missing. I assume this is a mistake?

Best,
Florian

ShawMerlin, 2024-02-07:
My schematics are a mess and I am working on getting them cleaned up and will upload them.

Also, with the 3.0 design being tested, I would not recommend using any previous boards.  If you have already put money/time into getting boards made, send me a DM and I will work on getting you a 3.0 board.

Thanks,
-Merlin Shaw

fschuetz, 2024-02-07:
> My schematics are a mess and I am working on getting them cleaned up and will upload them.
>
>
>
> Also, with the 3.0 design being tested, I would not recommend using any previous boards.  If you have [...]

Dear Merlin

Thank you for the quick reply. Do I understand correctly, that V3 its not yet released? Any ETA?

As for cleaning up the schema: Do you need help with that? I can happily work on cleaning up the schema (for the new version or the old version to have it for historic reason). Just let me know if you need help. 

Best
Florian

ShawMerlin, 2024-02-07:
V3 is not released but has been testing very well.

If you are interested in getting one of the prototypes, send me a DM and we
can work out price at cost only.
merlin.shaw at g-mail

Thanks,
-Merlin Shaw