ShawMerlin/NES-Power-Module-Redesign #15: Cannot power on if console left in Standby

Nidhoegger, 2024-02-25:
Hi, first of all, thank you for your work! I really love your mod!

One problem I am facing: I modded my consoles with this power board, since my old boards were defect. It works fine if plugged in an turned on directly. But if I leave the console turned off with the (original) Power Brick plugged in and want to play, the console does not turn on. I need to  decharge the capacitor or remove power and wait ~10 Minutes, then plug it in and it will turn on.

Is there a fix for that? Did I mess anything up? That happens for 100% of the boards (I revived about 5 consoles with it, all have the same issue).

Its the Gen9 board btw.

Thank you very much!

ShawMerlin, 2024-02-25:
Thank you for reaching out.

I have not heard of this issue before and have not seen it personally on
any of the consoles that I have used the RGB board on.  I have had had an
older board in my console for over 2 years now and it always stays plugged
in.

Have you tried a different power adapter for this issue?

What version of the power boards have you been using?

Thanks,
-Merlin Shaw

Nidhoegger, 2024-02-25:
Hey there, thank you for the answer. The board has the Datecode 4/26/2023. I tried with 3 different Powerbricks, happens on all.

Bestm
Nidhoegger

Nidhoegger, 2024-02-25:
Hey, I will do that in the comming week and get back to you. As for the mail, it seems like its censored by github :).

ShawMerlin, 2024-02-26:
Merlin.Shaw et google

Nidhoegger, 2024-02-26:
So first measurement are taken. On Pin 1, which should be 9V out, I get a reading of 16V DC. When turning the NES on, it drops to 14,14V DC on PIN 1 and I get the same reading on PIN 2, PIN 3 has a steady 4,991V DC. Ill report back with the values when the NES wont turn on. I use the offical Power Brick that is rated a 9VAC with 1.3A.

ShawMerlin, 2024-02-26:
Those measurements are good for the working example. I am very interested
in how it differs when you have it not working.

Thanks again,
-Merlin Shaw

ShawMerlin, 2024-02-27:
> So far, the error did not yet manifest, but I had a power outage tonight,
> so it was not permanently plugged in. What I can say for now, Ive turned it
> on 6 hours ago again, the Voltage in Standby on pin 1 so far climbed to
> about 17.9V AC... Directly after plugging in its about 16V DC.

You are seeing AC on pin 1?

What is the voltage rating of the 2200uF C1 capacitor that you have these
boards?

Nidhoegger, 2024-02-27:
25V

ShawMerlin, 2024-02-27:
The 16-17 volts DC are not alarming as your power brick for the NES is
unregulated and has no load on it while the NES is powered off.

With having a 25V capacitor, no voltage ratings are being exceeded so
everything so far is acting normal.  Hopefully it will go into this fault
condition so we can see what those voltages are at that moment.

Thanks,
-Merlin Shaw

Nidhoegger, 2024-03-02:
So, the funny thing is: I could not reproduce it the whole week. until I found out what is different. I have a setup with a NESRGB and its hooked up via a RetroVision HD Genesis 2 cable to a Retrotink into a HDMI Multiplexer into a Flint 4KP capture card. When I tested it this week, I did not have Video plugged in, I just had a box sitting here with only the power plugged in and nothing went wrong. Now Ive had the Video Setup hookes as well, and it acutally reproduced after one night! The underwhelming thing: I get the same reading on the pins... So I guess its nothing wrong with the power board, it is something else. It happenes on all my boxes. I dont know if the activly powered HDMI Multiplexer may inject some voltage into the board. Ill hook up my Oscilloscope later and check all the signals and pins... Its really strange.

ShawMerlin, 2024-03-02:
Thank you for your detailed analysis and your discovery on this.

What retrotink model do you have?  I would like to reproduce this here.

Thanks,
-Merlin Shaw

Nidhoegger, 2024-03-02:
I have a RetroTink 5X. And the HDMI Thingy is some cheap one off of amazon with no brand, but its activley powered.

ShawMerlin, 2024-04-03:
Since is the first report out of 100's of boards in the wild.  We will save this thread for future reference.

closed 2024-04-03 by ShawMerlin