ShawMerlin/NES-Power-Module-Redesign #16: Do you have a more detailed BOM for the USB-Powered-RCA-Classic?

AlbertVDS, 2024-03-31:
Do you have a more detailed BOM for the USB-Powered-RCA-Classic?
I don't seem be able to find the correct parts, or even parts for it at all.

ShawMerlin, 2024-03-31:
I can look at that to see if the parts can be more easily identified.

Although, we are about to finish a new USB C PD Power Board that will be a
lot better than this board.
The biggest issue with USB C is it allows for 4.75v to 5.75 volts and is
still considered in spec.

The NES does not like voltages under or over 5.0 - 5.1 volts so it's very
hit and miss with all the different power bricks out there.
With the USB-C PD design, we are hitting 5.07 volts perfectly, even under a
2 amp load.

Hoping to have this ready for release in the next 4 weeks,  you can shoot
me an email if you would like more information on this.

Thanks,
-Merlin Shaw

AlbertVDS, 2024-04-03:
Would be great if you could, because I already have the PCBs. 😅
Though if you think it's a better choice to just get the update version, then I might just do that.

ShawMerlin, 2024-04-03:
I still have extra parts for these boards.  Please send me an email and I will provide the parts for you.

ShawMerlin, 2024-04-06:
I have added website links to the parts on the BOM list.  I checked LCSC and all parts are available.

closed 2024-04-06 by ShawMerlin