ShawMerlin/NES-Power-Module-Redesign #18: LumaCode PCB Design

Nibell, 2024-07-27:
Hi,
I was think it would be neat if a new PCB existed to support LumaCode.
Saw the KiCad files are not uploaded, so thought i will create the next best thing.
Was thinking like this, sorry about my poor paint skills.
[image]
[PPUdigitizer for LumaCode](https://www.tindie.com/products/c0pperdragon/ppudigitizer-pal-only/) does not yet support NTSC, but will in the future.
Any ideas about that?
I happy to help if you need it.

Thanks,
-Linus Nibell

ShawMerlin, 2024-07-27:
Linus,

That you for your comment and I wanted to let you know that we have been working on something just like this.  :)

The Duo Mono Audio output has always been sort of a gimmick as you can simply just get a RCA splitter to use on the Red Audio Port and get the exact same results as what we are doing on the current board.

I have been wanting to move the Yellow port back to the original position for a long time now, especially for how awkward it looks to have the RCA cable jacks so far separeated when plugging it in.

We just did not have a use for that old RF output port but when I watched the video on Luma Code, which is pretty awesome!!, I had the exact same thought that you did.  :)

It is super easy just to add a large pad that folks can solder any type of external video on, like lumacode, and have it output what they desire.

My plan was to have this implemented on the Classic RCA editions for release 3.2 and hopefully it will be ready in a week or so.

Thanks,
-Merlin Shaw

Nibell, 2024-07-27:
No problem Merlin,
Saw it this week also, i will gladly test any prototype board. Don't have the PPUdigitizer yet, waiting on it to ship.
Live in Sweden so have access to pal consoles, they are pretty cheap here so just pm me and i can send you one 😄

ShawMerlin, 2024-07-27:
If you can reach out to my email.  merlin dot shaw at google then we I can list you as a tester and get you some prototypes.  :)

closed 2024-11-04 by ShawMerlin