ShawMerlin/NES-Power-Module-Redesign #19: Gerbers for 3.1c don't appear in repo?

hexbus, 2024-07-29:
https://github.com/ShawMerlin/NES-Power-Module-Redesign/tree/main/Classic%20Composite%20Edition doesn't appear to have any Gerber's in the repo that I can locate; do you plan on making these available, or are they located in a different directory structure?

Cheers,
JD

ShawMerlin, 2024-07-29:
Thank you for pointing this out, I will get those uploaded later today.

hexbus, 2024-07-31:
Thank you so much!  I also wanted to let you know how much I appreciate you redesigning these for the community!  I have my own childhood NES and a couple of other friends who I will be using these for to clean the image up on (not to mention - the power).  You have done an amazing job on these - they are very well laid out.  

I wonder if I could shoehorn one of your AV models in my Famicom too - should work if I soldered it I would think.  (I might do it just to see it work, but it won't fit permanently) :)

ShawMerlin, 2024-11-04:
The Gerber files and assembly files have been added to the link above.  Thank you again for bringing htis to my attention.

The board will work in a famicom but with the size differences, it would be quite a retrofit.  There are famicom module replacements out there.

Thanks,
-Merlin Shaw

closed 2024-11-04 by ShawMerlin