ShawMerlin/NES-Power-Module-Redesign #21: Use gen2 module without NESRGB possible?

danyboy666, 2025-01-11:
I have the version 3.1c and I can't get it to show anything other than a vertical and horizontal white line? The board i want to use is an opentendo which is 100% populated with componants i'm sure are all good. CIC, CPU and PPU all pulled from a working nes. So my question is, is there a way to hook the av module without an nesrgb mod?

ShawMerlin, 2025-01-11:
If you do not have a RGBNES, you can still get audio out through the AV
with the selector in the internal video position.

This board doesn’t do any video processing so if you aren’t seeing a
picture, there may be an issue elsewhere.

What you can do to troubleshoot the board is look at the voltage on the top
of the 5 interconnect pins.  It should be 12volts or higher, then check the
2nd pin with the power switch on and it should be also 12 volts or higher.
Then check the 3rd pin and you should have 5 volts on that pin.  Let me
know if you are able to test these points.

Thanks,
-Merlin Shaw

danyboy666, 2025-01-11:
Alright I will let you know. I don't think it's a voltage issue I think i've narrowed down to bad ram. I have a stack of 6116 they might be the issue along with a suspected bad ppu. I have a couple sets of cpu ppu along with good ram chips. I will do some more testing tomorrow i've just finished desoldering ram sockets from a working nes pcb for testing. I'm sure it's an issue on my side thanks for your great work on this I have printed several from jlcpcb already just wanted to make sure it would work without rgb mod board. Thanks alot.

danyboy666, 2025-01-11:
In your experience is it possible for a bad ppu to corrupt/break a ram chip? I've have a couple of brand new 6116 equivalents but all.seem to output the same lines on screen.

ShawMerlin, 2025-01-11:
I have never seen a ram chip impacted by the PPU but anything is possible.

danyboy666, 2025-01-12:
You were right. CPU/PPU are all fine. I have two sets of them. I did have an issue with a ram chip but wasn't the ones I was using for the opentendo board. I did test the voltages and like you say I get 12v on two pins, 5v on the middle pin. Switches are set to internal vid/audio. I might have to swap tge power module for an original one just for testing. I really doubt it's one of the passive componants. [image]

[image]

[image]

[image]

[image]

ShawMerlin, 2025-01-12:
Are you using an actual CIC chip or one of the open source replacements?

I have heard of some of the open source CIC solutions having issues with
the rise of the buck converter which require a few different sizes of
capacitors to fix.
Let me know if that is the case for you and I can see if I can find that
info on what to try.

Thanks,
-Merlin Shaw

danyboy666, 2025-01-12:
Yes it is the official cic which I confirm is working. I've also saw somewherenon reddit a user having luck with swapping C44 for something lower. Altough 220pf is the same value as on the original board. Also i've tried hooking up a chinese rgb mod board. I do get some garbbled characters and lots of static on screen using composite only. Will try to post some pics. I've used an original trimmer cap as well as the original nes oscillator. All does the same output on screen.

ShawMerlin, 2025-01-12:
The YouTube channel “More fun making it” has a video of him using the
RGBNES and one of these new power boards.  He ended up having to change
capacitor values too but i think that was more related to the RGBNES
board.  It might be helpful so I thought I would share.

If you have access to a original power module, you could swap it in to take
the power board out of the equation.

Thanks,
-Merlin Shaw

danyboy666, 2025-01-12:
Yes I think that would be the logical step to try to rule out anything related to the power module at least. So close yet so far lol. I'll try this tomorrow I do have acces to a power module. I even have a second opentendo board pretty much ready to test. Thanks alot for this.

danyboy666, 2025-01-13:
It only took three days of screwing around.... I should have tested with and original power module at first. [image]

[image]

It's working even without the trimmer cap. Something is up with the power module I was sold I'm not sure what. I think i'll retest it just to be sure but it's almost certain it's that. Maybe a bad solder somewhere or even a bad (only)  transistor. I did check it to be sure with a diode tester but in circuit so i'm not entirely sure it's right or not but i'm not reading the same values. I'm kinda glad it wasn't my opentendo build lol.

danyboy666, 2025-01-13:
If I test with the new power module, problem is back.[image]

ShawMerlin, 2025-01-13:
> If I test with the new power module, problem is back.20250113_130703.jpg

The issue comes from a buck converter has a gradual rise to 5 volts while
the original Power Module is a LDO so it's instant at 5 volts.
This would be the same case for any switching buck converter providing the
voltage to the main board.

We have not ever seen this on the original main NES Board so there is
something with how the timing circuit is working from the new main board.
I have heard of others with OpenTendo's that had this issue with the new
power module and they had to find the right capacitor in order to get it to
work.

Too low of a capacitor or none at all means that the timing hits too fast,
too large of a capacitor means the timing hits too slow.
From my notes, it looks like the last person that had issues used a 10uF
capacitor to get his Opentendo to work with the new power module.

If the original power module is working for you, there isn't a huge impact
with using it compared to the new switching power module.

Thanks,
-Merlin Shaw

danyboy666, 2025-01-13:
I do have some plans with those boards so idealy would be nice to have them work on opentendos since I have some main board pcbs already printed. Do you remember which capacitor? If it's the timing circuit my guess would be C44 like i've mentioned before? It would be certainly helpful to others if they have this issue too. At least it would be documented somewhere.

ShawMerlin, 2025-01-13:
I believe it was the  10uF capacitor that he used.

There is an active discussion on it right now with the Dubesihower's
Modding group under NES-Help
https://discord.gg/MUc7uMyH

Zaxour is on that thread and has a lot of information on the timing circuit
and these capacitors.
I am sorry that I am not as much help as with analog things, timings are
very critical for the reset circuit.

I plan on documenting it on Github when there is more of a concrete answer.
Lots of Opentendo's have worked without having to mess with the timing
capacitor, only just a couple of shown this issue.

Thanks for your patience on this,
-Merlin Shaw

ShawMerlin, 2025-01-15:
Thank you for catching it,  For Digikey it should be 2SA1015 or KSA1015YTA.  Both will work.  I will get that updated right now.

danyboy666, 2025-01-15:
Ohhhh so I do have the wrong one installed? It says LG 416 on the package and tested as a NPN. If you're positive it an ASA2015 i'll swap it out right away :)

ShawMerlin, 2025-01-15:
Did you get this board from me on my eBay?  the JLCPCB number C181170 is correct so if these are ordered from JLCPCB, then it would have come with a 2SA1015 - https://jlcpcb.com/partdetail/GuangdongHottech-2SA1015/C181170

danyboy666, 2025-01-15:
> Did you get this board from me on my eBay?  the JLCPCB number C181170 is correct so if these are ordered from JLCPCB, then it would have come with a 2SA1015 - https://jlcpcb.com/partdetail/GuangdongHottech-2SA1015/C181170

It was a seller named kurt. It came all pre soldered. 

Good thing now i'll be able to order the right parts for the boards I have here. Also there are two BOMs for the genesis 2 version and C12 is either listed as 1nF or 560pF. It's not a big difference but still. Might be a good time to validate that too.

ShawMerlin, 2025-01-15:
I will check on C12.  I do apologize for this mistake of the Digikey part number for Q1.  As more people start offering these, having accurate information will be critical for our community to get working safe devices for their legacy hardware.

I have a lab bench with a frankenstine'd NES and pogo pins that I use to test all the boards I have sold.  I have them come pre-assembled from JLCPCB with the SMD parts so I would not have caught the DigiKey part number mistake.

Thank you again for discovering that and bringing it to our attention.
-Merlin Shaw

ShawMerlin, 2025-01-15:
For C12;
 
Both 1nF and 560pF will work without issue.  It will work with this removed too.  The TinyNES's design added the 560pF so we changed ours to this and it does help make a more crisp picture.

So I would recommend 560pf in this location.  I will make sure the BOM's reflect this.

danyboy666, 2025-01-15:
> For C12;
>  
> Both 1nF and 560pF will work without issue.  It will work with this removed too.  The TinyNES's design added the 560pF so we changed ours to this and it does help make a more crisp picture.
> 
> So I would recommend 560pf in this location.  I will make sure the BOM's reflect this.

Oh very nice i'll grab that new version. Thanks alot for this project. It's a board which will be very uselful to me. Not much else is needed to build an entire nes.

ShawMerlin, 2025-01-15:
Here is what i'm updating the BOM with - https://www.digikey.com/en/products/detail/kemet/C0805C561J5GACAUTO/10642381

danyboy666, 2025-01-17:
So i'm guessing by the non-answer the problem is on my side and I should deal with this on my own? Man i've spent almost a 100$ on this module that should have worked. I'll leave the rgb alone for now but still that tone is really annoying.

ShawMerlin, 2025-01-17:
I do apologize as I did not see your follow up question.

The buzzing on the sound was something that was discovered before 3.1c ever
was released or put into production.

It’s due to the buck converter itself and it’s a variant that causes this
buzzing due to its claims to reduce EMF.

If you can get a nice close up picture of the buck converter,  you should
see if their is an H or not in the part number.

If you can reach out to me on my person email.  Check out my GitHub file on
the GND loop for my address.

I will get you taken care of, i didn’t sell you the board but it was built
using my project and possible an older BOM (I will verify that too)

Thanks,
-Merlin Shaw

danyboy666, 2025-01-17:
> I do apologize as I did not see your follow up question.
> 
> The buzzing on the sound was something that was discovered before 3.1c ever
> was released or put into production.
> 
> It’s due to the buck converter itself and it’s a variant that causes this
> buzzing due to its claims to reduce EMF.
> 
> If you can get a nice close up picture of the buck converter,  you should
> see if their is an H or not in the part number.
> 
> If you can reach out to me on my person email.  Check out my GitHub file on
> the GND loop for my address.
> 
> I will get you taken care of, i didn’t sell you the board but it was built
> using my project and possible an older BOM (I will verify that too)
> 
> Thanks,
> -Merlin Shaw
> 
> 
> 
> > So i'm guessing by the non-answer the problem is on my side and I should
> > deal with this on my own? Man i've spent almost a 100$ on this module that
> > should have worked. I'll leave the rgb alone for now but still that tone is
> > really annoying.

I know it's not your problem also I really didn't want to come accross as an a**. I'll post some pics thanks again for your help. I really do apreciate it.

danyboy666, 2025-01-31:
Everything is solved. Thanks for everything sorry to see you retire this project.