ShawMerlin/NES-Power-Module-Redesign #26: BOM for Classic Composite Edition not correct

sureal808, 2025-04-17:
R2 references 120k ohm resistor but links to a 105k resistor. Which is the correct one?

sureal808, 2025-04-17:
R1 does not have a link. Should link here: https://www.digikey.ca/en/products/detail/vishay-dale/CRCW08054K70FKEAC/7928553

nebhead, 2025-04-21:
Also, the switch listed in the BOM (SW1 | Switch | E-Switch |   | DPDT | EG2215) doesn't seem to have the correct footprint, or it's the wrong switch.  At least for the Classic THT DIY edition.

nebhead, 2025-04-21:
> The switch did change on a few boards and not everything got updated
> correctly.  To make sure we cover all basis, I can add both switches to the
> BOM.  It would have been better if I changed the SW tag so they didn’t
> match.

Thanks for the super fast response!